RoelKroes/TBTracker #12: no matching function for call to 'SX1278::beginFSK(float&, float&, float&, float&, int8_t&, uint8_t&, uint16_t&, bool&)'

FoxnoTango, 2020-08-30:
Hello Roel,

When compiling the code in my IDE I get an error message:

no matching function for call to 'SX1278::beginFSK(float&, float&, float&, float&, int8_t&, uint8_t&, uint16_t&, bool&)'

RADIO:60:53: error: no matching function for call to 'SX1278::beginFSK(float&, float&, float&, float&, int8_t&, uint8_t&, uint16_t&, bool&)' FSKSettings.EnableOOK);

I have installed the libraries RadioLib and TinyGPS++.
I have tried to compile this code on two different systems but they both gave the same result.
What could be the problem?

Foeke.

RoelKroes, 2020-08-31:
Hi Foeke,

The Radiolib library has been updated and this causes the problem in the
code.
I am in the middle of a "home improvement" phase and had no time to update
the code. But I will do so in the near future.
For now, you best bet will be to use one of the older versions of the
Radiolib library, which you can install using the Arduino Library Manager
(Ctrl+Shift+I).

Hope that works for you!

Roel.

roel@kroes.com

RoelKroes, 2020-09-19:
This has now been solved.

closed 2020-09-19 by RoelKroes