RoelKroes/TBTracker #15: Radio.ino: error: 'ERR_NONE' was not declared in this scope

star210, 2022-01-06:
Hi 

I am trying to compile the code for a Atmega328PU and get the following error:

Radio.ino: In function 'void SetupFSK()':
Radio:63:15: error: 'ERR_NONE' was not declared in this scope

So when there is no error it will display the debug text to the serial monitor but I cant see where ERR_NONE is defined?

Am I missing something? I have both recommended librarys installed. 

Thanks

RoelKroes, 2022-01-07:
Hi,

It seems that the status codes in the latest radiolib library have updated.
Try changing ERR_NONE to RADIOLIB_ERR_NONE and let me know if that works for you.

Roel.

star210, 2022-01-09:
Hi 
I changed to RADIOLIB_ERR_NONE as suggested and it now compiles.
Thankyou Roel.

RoelKroes, 2022-01-12:
Thanks for reporting it, I will change the code for this.

Roel.

closed 2022-01-12 by RoelKroes