RoelKroes/TBTracker #16: TBTracker.ino not compiling

Sayed-Belal, 2022-03-30:
Hey. When I copied and pasted the TBTracker.ino code onto my Arduino IDE and also included the RadioLib and TinyGPSPlus libraries, there is an error compiling and the compiler says "'ResetRadio' was not declared in this scope". Please help me with this.

Here is my code:
https://docs.google.com/document/d/1rnmXTs5lt-sbBCd-BmCGBPNFCoarGlDjtIVzv_ftaZM/edit?usp=sharing

RoelKroes, 2022-03-30:
You cannot copy and paste one single file into the IDE. You need to:

1. download ALL files (as a zip file) by pressing "Code" and then "Download ZIP"
2. unzip all the files
3. put all files in a folder named "TBTracker"
4. open tbtracker.ino in the IDE 
5. compile

(ResetRadio is a function that is declared in radio.ino. That file must be in the same folder as tbtracker.ino. It is not enough to only copy and paste tbracker.ino.

Hope this works for you.

Roel.

Sayed-Belal, 2022-03-30:
Perfect. Done. Thank you!