RoelKroes/TBTracker #19: Adding Field List to allow Sondehub support

KevWal, 2022-09-25:
Hi Roel

A placeholder for fieldlist support to allow Sondehub to understand extra data as per:

https://www.daveakerman.com/?p=2987

I may have a look at implimenting in the comming weeks

Cheers
Kev

RoelKroes, 2022-09-25:
Hi Kev,

Thanks for letting me know!  
I've seen that and indeed an excellent idea from Dave,

I already implemented the code for that last week and tested it. I will release the code this week.
I also updated the code to be fully compatible with TinyGS.
Luckily a minor change, only like 5 lines of code, so no memory problems :-)

Best
Roel.

KevWal, 2022-09-25:
Fantastic thanks Roel, looking forward to testing it when you are ready!

Cheers
Kev

RoelKroes, 2022-09-27:
I Added the support for (optional) transmitting METADATA compatible with lora-gateway from Dave Akerman and Sondehub. 

Best,
Roel.

closed 2022-10-05 by RoelKroes