RoelKroes/TBTracker #21: Problem with compiling Arduino IDE 2.0.x  on Windows and Linux

edegraaff, 2024-01-01:
Compiling the code on Windows 11 and the latest Arduino IDE 2.2.1 causes problems with the ArduinoPro-Mini (5v 16mhz) during compilation. Memory error with statement take out comments to get it fit.

I disabled the Lora module and only enabled the RTTY module and debug. Even then 
Sketch uses 35888 bytes (116%) of program storage space. Maximum is 30720 bytes.
Global variables use 1291 bytes (63%) of dynamic memory, leaving 757 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board

Even switching off devmode does not solve the issue in Windows environment. 

The issue is also on Ubuntu

edegraaff, 2024-01-01:
Solution Roel suggested is to use the older RadioLib 5.1, and this did the job. 
It seems there is a way to free up memory by using an external programmer as decribed here: 
https://docs.arduino.cc/hacking/software/Programmer?queryID=undefined

RoelKroes, 2024-01-02:
We've seen this problem before. I've added the solution to the readme page. Thanks for reporting!

closed 2024-01-02 by RoelKroes